ros-industrial/abb_librws #3: Enable non-ROS builds

gavanderhoorn, 2018-04-04:
Adapt `CMakeLists.txt` so we can accomodate building the library without Catkin / in a non-ROS (build) environment.

gavanderhoorn, 2019-08-23:
Edit: as there is no ROS-specific functionality in this package, we should migrate it to be a pure CMake package.

The manifest can remain, and this would allow us to still build it in a ROS workspace (by setting [build_type](https://www.ros.org/reps/rep-0127.html#build-type) to `cmake`).

traversaro, 2019-08-23:
> Edit: as there is no ROS-specific functionality in this package, we should migrate it to be a pure CMake package. All `catkin_*` calls should be removed.
> 
> The manifest can remain, and this would allow us to still build it in a ROS workspace (by setting [build_type](https://www.ros.org/reps/rep-0127.html#build-type) to `cmake`).
> 
> (this would also immediately make non-ROS builds possible, without needing two build scripts or undesirable detection/if-else constructs)

That is great. Just to understand, the same applies to `abb_libegm`? If that is the case, what changed w.r.t. to the requirement of being able to build the package with `catkin_make` mentioned in https://github.com/ros-industrial/abb_libegm/issues/3#issuecomment-409187490 ?

gavanderhoorn, 2019-08-23:
> Just to understand, the same applies to `abb_libegm`? 

Yes.

> If that is the case, what changed w.r.t. to the requirement of being able to build the package with `catkin_make` mentioned in [ros-industrial/abb_libegm#3 (comment)](https://github.com/ros-industrial/abb_libegm/issues/3#issuecomment-409187490) ?

ROS 2 and the existence of Colcon.

Additionally, `catkin_tools` may be used with ROS 1.

jontje, 2019-11-11:
I have created a new [branch](https://github.com/jontje/abb_librws/tree/plain_cmake) for converting the package into a plain cmake package.

I.e. similar rationale as in this comment https://github.com/ros-industrial/abb_libegm/issues/3#issuecomment-552378566.

gavanderhoorn, 2020-05-20:
Looks like #69 was merged, so closing.

Thanks @traversaro  and @jontje :+1:

closed 2020-05-20 by gavanderhoorn