ros-industrial/abb_librws #8: Ship a Poco find script that supports checking versions

gavanderhoorn, 2018-04-05:
The `FindPoco.cmake` provided by `cmake_modules` doesn't set the required variables for CMake to actually verify that it found the requested version of `libpoco`. At the moment the `find_package(Poco .. 1.4.3 ..)` is thus not enforced. This can lead to build issues on systems where `libpoco-dev` is installed, but not the right version.

This should be fixed by adapting `FindPoco.cmake` to also extract the version of the found `libpoco-dev`, and then shipping that version of the find script with the sources.

gavanderhoorn, 2018-04-05:
The `FindPoco.cmake` in `ros2/poco_vendor` ([this one](https://github.com/ros2/poco_vendor/blob/c545f401cdec96007d18ba5a096acb0fff1c4b6c/cmake/Modules/FindPoco.cmake)) does some checking for versions, but does not appear to follow CMake's way of doing things (storing the version in a `Poco_VERSION` variable).

This could possibly be adapted.

de-vri-es, 2019-09-02:
Note that poco ships its own `PocoConfig.cmake`, which is always preferable over a custom `FindXXX.cmake`. So there really should be no need for this.