ros-industrial/abb_librws #20: Expand the readme

jontje, 2018-09-06:
Add sections for:
- Structure/class descriptions
- Troubleshooting
- Acknowledgements

gavanderhoorn, 2018-09-06:
See my comment here: https://github.com/ros-industrial/abb_libegm/issues/17#issuecomment-419027734.

jontje, 2018-09-06:
That sounds reasonable, see my reply here https://github.com/ros-industrial/abb_libegm/issues/17#issuecomment-419059317.

Except that the  "structure/class descriptions" would be something like this for RWS:
- Add a simple illustration of how the **important classes** are related to each other.
- Add some very brief lines for each class. For example:
  - **POCOClient**
_Sets up, and manages a HTTP client and (if needed) a WebSocket instance (based on POCO classes)_
  - **RWSClient**
_Expands the POCOClient for using RWS (e.g. exposing RWS services and resources, evaluating the communication results)_
  - **RWSInterface**
_Provide more user friendly APIs (e.g. wraps a RWSClient and provide simpler methods)_
  - **RWSStateMachineInterface**
_Extends the RWSInterface for a system specific case (i.e. requires specific configurations and RAPID program running on the robot controller)_

This just so that one can get a quick overview, without digging into the details.

jontje, 2019-03-12:
**Regarding troubleshooting:**
The readme should be update to include a description of the problem mentioned in issue https://github.com/ros-industrial/abb_librws/issues/47 and the possible solution in comment https://github.com/ros-industrial/abb_librws/issues/47#issuecomment-471869289

gavanderhoorn, 2020-05-20:
> **Regarding troubleshooting:**
> The readme should be update to include a description of the problem mentioned in issue #47 and the possible solution in comment [#47 (comment)](https://github.com/ros-industrial/abb_librws/issues/47#issuecomment-471869289)

Tracking this outstanding task in #81 (which I believe is a nicer solution than the proxy approach described in https://github.com/ros-industrial/abb_librws/issues/47#issuecomment-471869289).

gavanderhoorn, 2020-05-20:
Closing, as I believe all the other additions to the readme have been made.

Further improvements can be suggested/tracked in other issues.

Thanks @jontje :+1:

closed 2020-05-20 by gavanderhoorn