ros-industrial/abb_librws #21: Create code samples

jontje, 2018-09-06:
Create samples to showcase how to use the interface classes, including:
- C++ code
- RAPID code (and possibly a RobotStudio Pack&Go file for a simulation setup)
- Instructions

gonzalocasas, 2019-02-24:
Yes, please. Without at least a minimal example, I find it extremely hard to use this library.

gonzalocasas, 2019-02-25:
@jontje thanks a lot! yes, this helps a lot!

rsaltus, 2019-05-30:
Hey @jontje, I'm having trouble implementing the abb_librws_samples.zip file you've included in the thread above, maybe you can help out. I'm trying to connect to a Windows machine running RobotStudio virtual controller with RW 6.09 from an external laptop running Ubuntu 14.04 and ROS Indigo. I've tried all of the example files you've included, but cannot get any response from any of them. I've attached a sample output (with IP censored) from the a1_collect_info_node:  

> [ INFO] [1559253131.475575020]: Using provided IP-address: "---.--.---.--"
[ INFO] [1559253131.475679247]: ========== Collecting info sample ==========
[ INFO] [1559253131.475693087]: 1: Press Enter to collect information
>
>[ INFO] [1559253132.530105635]: Auto mode: unknown
[ INFO] [1559253132.532249484]: Motor on: unknown
[ INFO] [1559253132.546998364]: RAPID running: unknown
[ INFO] [1559253132.561179831]: Current jointtarget: [[0,0,0,0,0,0],[0,0,0,0,0,0]]
[ INFO] [1559253132.563248354]: Current robtarget: [[0,0,0],[0,0,0,0],[0,0,0,0],[0,0,0,0,0,0]]

I'm pretty sure the RWS Server is running fine on the Windows side, I followed the steps at http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/index.html under "Discovery" and received an IP and proper node name. I can also ping between the two computers, and have already established a working EGM connection. 

Right now I've tried to implement this using both a barebones VC running RW 6.09 with the sample RAPID code you provided in the abb_librws_samples.zip file, and another VC running the StateMachine Add-In separately. Neither gets any response. One thing I wasn't sure about: do I need to do any setup on the RobotStudio side in order to allow the RWS to access the VC data?

If you have any suggestions, they'd be greatly appreciated.

gavanderhoorn, 2019-05-31:
This is most likely the issue where RWS won't accept requests from a non-local IP. See #47.

jontje, 2019-05-31:
Hi @rsaltus,

I think @gavanderhoorn is correct.

Just to verify the issue, then you can try the following:

- In the Windows machine:
  1. Open a web browser
  2. Go to http://127.0.0.1/rw?debug=1
  3. Log in with "Default User" and "robotics"
  4. If the RWS server is up and running, then you should get a list of RobotWare services

- In the Ubuntu machine:
  1. Repeat the process above (but replace 127.0.0.1 with the Windows machine's IP)
  2. If you get the response "RAPI Unidentified Error" it means that you connected to the RWS server, but the request was not accepted

As @gavanderhoorn mentioned, you can try the workaround suggested in https://github.com/ros-industrial/abb_librws/issues/47

rsaltus, 2019-06-11:
Thanks for your help @gavanderhoorn and @jontje, the issue was exactly like you guys thought. I implemented the proxy server workaround in #47 and it worked like a charm.

lijianhualeigua, 2019-09-26:
Hey @jontje  , i download the abb_librws_samples.zip, but not find the  RobotStudio Pack&Go file RWSSamplesSolution.rspag  in  abb_librws_samples file , can you  reupdate this abb_librws_samples.zip. Thanks a lot.

jontje, 2019-09-27:
Hi @lijianhualeigua, here is an updated zip-file: [abb_librws_samples.zip](https://github.com/ros-industrial/abb_librws/files/3664009/abb_librws_samples.zip)

I have added a Pack&Go file based on RobotWare 6.08.01, as well as added some launch files for the different ROS nodes.

I hope this helps! Also, I am currently out-of-office so I might be slow to reply to questions.

lijianhualeigua, 2019-09-29:
Thanks for your help @jontje .

manuelko, 2020-10-18:
Hello everyone. I do not know if this is the right place to ask?
Here it comes: I do not understand why the robot controller must be in AUTO mode to write to a RAPID variable in example "b1_rapid_data_node"? Is there a way to write to a variable in Rapid, in Manual mode? I can not find any answer after spending hours trying to solve this problem.
Here is my setup:
IRB4600
RobotWare 6.10.02
Hope someone has time to respond to such a relatively old thread.

jontje, 2020-10-19:
> I do not understand why the robot controller must be in AUTO mode to write to a RAPID variable in example "b1_rapid_data_node"?

It isn't really needed to be in `AUTO` mode, the example assumes `AUTO` mode since it is easier to test against.

> Is there a way to write to a variable in RAPID, in Manual mode?

Yes, but you must request mastership over the `RAPID` resource or register as a local user.

Last time I tried the request mastership approach, then it didn't work so well. So you could try to use the [registerLocalUser(...)](https://github.com/ros-industrial/abb_librws/blob/3946fe736672b1a9213d6213318e215d0b78036b/include/abb_librws/rws_interface.h#L546-L548) method and then use [registerRemoteUser(...)](https://github.com/ros-industrial/abb_librws/blob/3946fe736672b1a9213d6213318e215d0b78036b/include/abb_librws/rws_interface.h#L559-L561) to unregister.

manuelko, 2020-10-19:
Thank you very much for the answer, it is very much appreciated.
After many hours, I have come to the same conclusion.
I have changed this in the b1 example and I can access with the new user. However, the controller behaves the same way as before. I think maybe IRC5 will not let go of ownership when running in Manual mode. 
In a way, it makes sense, however, very impractical when I want to send sensor data to a variable through ROS.

`   abb :: rws :: RWSInterface rws_interface (rws_ip_address, "ROS_Machine", "random_password");
   rws_interface.registerLocalUser ("ROS_Machine", "ROS", "Lund");`

Have also tried with `registerRemoteUser`, without success

blakesba, 2021-11-30:
In trying to run the samples, I am getting an error about a header file that is imported in rws_common.h, but that I do not see in the repo - abb_librws_export.h. The error is:

fatal error: abb_librws_export.h: No such file or directory

Is there something that I am missing in the setup or another dependency that needs to be installed that contains abb_librws_export.h?

Thank you.