ros-industrial/abb_librws #47: 500 : Internal Server Error

BidoMan, 2019-03-10:
Hi,
I have a problem and would appreciate any help, Whenever i perform a request, for example like the below code snippet :

`
#include "abb_librws/rws_state_machine_interface.h"
using namespace abb;
using namespace rws;
using namespace std;

int main()
{
string ip_address = "192.xxx.xxx.xxx";
RWSStateMachineInterface IRB_120(ip_address);
TriBool State;
State = IRB_120.isRAPIDRunning();
return 0;
}
`
I get 500 Internal Server Error 
**here's the output from wireShark:**
[image]

I have basic robotstudio version 6.07, robotware 6.08.01
I am using the state machine Add-In.

gavanderhoorn, 2019-03-10:
Looks like you first get a `401 Unauthorized`. Then a `500`.

BidoMan, 2019-03-10:
yes, but i thought this was necessary , as i understood from abb's api center, Authentication section:

http://developercenter.robotstudio.com/webservice/api_reference

[image]

jontje, 2019-03-11:
@BidoMan: I am assuming that you are using RWS with a virtual controller, and that you are trying to make RWS requests from an external computer/environment to a computer running a RobotStudio simulation.

**Firstly**, it is correct to get a `401 Unauthorized` reply when initiating an RWS communication session, so that is not a problem.

**Secondly**, if you have a setup similar to my assumptions, then the RWS server will block such requests and reply with `500 Internal Server Error` and this behaviour is due to ABB's cyber security policies. For example, to prevent access to the file system of the computer running RobotStudio (i.e. via RWS's fileservice feature).

This is just a rough idea, and there are probably better solutions, but you could bypass that behaviour if you for example implement a simple HTTP proxy server, on the computer running RobotStudio, that only allow certain RWS requests. The proxy server could take RWS requests (on another port number), make the request (locally) to the RWS server, and send back the reply. 

Also, this is not an issue when using RWS with a real robot controller, I have only seen it occur with virtual controllers.

BidoMan, 2019-03-11:
Yes, you are right, i am using a virtual control in another computer running RobotStudio.

So can i make this proxy server from inside robot studio? because i have seen this option in the controller configuration section :

[image]

BidoMan, 2019-03-11:
Ok , so i searched for how to do this proxy server, and there are several applications that can be used to create a proxy server like CCproxy on windows, and if i understand right, i will use the provided controller configuration in RobotStudio to communicate with this proxy server i created.

jontje, 2019-03-11:
The `Connected Services` configuration is used to connect to ABB Connected Services Center and it is not related to what I was writing about. You shouldn't actually need to make any changes in the configurations.

What I meant was to implement a proxy server in for example C# (possibly as a [RobotStudio Add-In](http://developercenter.robotstudio.com/blobproxy/devcenter/RobotStudio/html/08a465af-bdcc-4f65-a647-764c14005d00.htm)) that basically sets up a `HttpListener` to listen for external RWS requests, and a `HttpClient` that locally relays the requests to the RWS server. The virtual controller would then only see requests made from the local client and not know about the external computer at all.

But I guess it is preferable if you can make use of something that already exists.

BidoMan, 2019-03-12:
Thanks a lot, it worked after i used a proxy server implemented in python, that was available  on github:
https://github.com/abhinavsingh/proxy.py

jontje, 2019-03-12:
Great! Then I assume I can close this issue, and I will add a comment to issue https://github.com/ros-industrial/abb_librws/issues/20 for including this in a troubleshooting section in the readme.

closed 2019-03-12 by jontje

j-polden, 2019-03-13:
Hi @BidoMan , I was wondering if you could please provide a few simple details on how you set up the proxy server you mentioned above. I'm currently trying to do the same so some pointers would be most useful. Thanks!

BidoMan, 2019-03-14:
ok here's what i did:

1- Setup python 2.7.12 (that's the one i used, don't know if newer versions will work) on the pc where you are running RobotStudio.
2- Clone the proxy server code  to your pc.
3- Run the code by:
        1-Run the cmd console.
        2-Go to the proxy server code folder using this command "cd /codepath".
        3-Run the code using this command:
                python proxy.py --hostname ipaddress --port 8080
           replace ipaddress with your pc's ip address, and you can change the port to any number you 
           want that is viable as a port number.
        4-Adjust the proxy settings using the ipaddress and port number you used above.
           if you are using Windows, follow this link to adjust proxy settings:
           [https://www.dummies.com/computers/operating-systems/windows-10/how-to-set-up-a-proxy-in-windows-10/](url)

That's it , just run your code that uses abb-librws and run RobotStudio in the other pc and you are good to go.
hope this helps.

lijianhualeigua, 2019-09-19:
Hi , I followed your steps of you mentioned above to  set up the proxy server  and found that I still can't connect to robotstudio. Why is that?
When i  running the python proxy.py  , theinput parameter of the  IP address I can only use 127.0.0.1, if  i use other IP addresses, the program will runing fail.
 For example: python proxy.py --hostname 192.168.0.173 --port 8080
The IP of the pc where running RobotStudio and  proxy.py  is 192.168.0.160.
The IP of the pc where running ROS is 192.168.0.173.
The error information:
`2019-09-19 20:55:18,583 - INFO - pid:20568 - load_plugins:1272 - Loaded plugin <class 'proxy.HttpProxyPlugin'>
2019-09-19 20:55:18,583 - INFO - pid:20568 - setup:282 - Starting 8 workers
2019-09-19 20:55:18,765 - ERROR - pid:20568 - run:247 - Exception while running the server gaierror(11001, 'getaddrinfo failed')`

Can you give me some pointers. Thanks a lot!

manuelko, 2020-04-14:
We had a lot of problems with the way to set up a proxy, so we found the link below. It works well.
https://forums.robotstudio.com/discussion/12082/using-robotwebservices-to-access-a-remote-virtual-controller

gavanderhoorn, 2021-02-22:
To make it more visible: #81 documents / references the correct / recommended way to solve this problem.