ros-industrial/abb_librws #49: EGM : Proportional Position Gain

BidoMan, 2019-03-12:
Hi,
In the document describing EGM motion (Position Guidance) and control loop there are two main control parameters that i can control, one is the Low Pass Filter the other is Proportional Position Gain, but i can't seem to find any way to control Proportion Position Gain while the code is running, is the only way to change it through hard coding in RobotStudio?, isn't there anyway to change it using abb_librws?

**Here is the part where it's mentioned:**

[image]

**Also is it the "K-factor" in these descriptions?**

[image]

Thanks.

jontje, 2019-03-13:
@BidoMan, this comment grew slightly longer than I intended... so enjoy :)

**Firstly**, the "Proportional Position Gain" is the same as the "K-factor", and it is also the same as the "Position Correction Gain" parameter of the RAPID instructions `EGMRunJoint `and `EGMRunPose`.

The "Low Pass Filter" parameter is set in the `EGMActJoint` and `EGMActPose` RAPID instructions, but I have not noticed any big effect of that when using the UDP variant of EGM. It is probably more important if you use the IO-signals variant of EGM, but I have not used that one myself.

Also, there is another parameter for the `EGMActJoint` and `EGMActPose` instructions called "Max Speed Deviation", which will have a big impact on the EGM motions. It sets the maximum allowed joint speed change in degrees/seconds and it is a factor to trim acceleration and deacceleration with.

**Secondly**, then abb_librws can be used to read and write RAPID variables, and this can be utilized to set parameter values that you have defined inside your RAPID program. 

The [RWSClient](https://github.com/ros-industrial/abb_librws/blob/master/include/abb_librws/rws_client.h) and [RWSInterface](https://github.com/ros-industrial/abb_librws/blob/master/include/abb_librws/rws_interface.h) classes has the methods `getRAPIDSymbolData(...)` and `setRAPIDSymboldData(...)` to facilitate this. These classes don't make any assumption on the RAPID program, so you will have to implement that part yourself.

However, if you are using the [RWSStateMachineInterface](https://github.com/ros-industrial/abb_librws/blob/master/include/abb_librws/rws_state_machine_interface.h) in combination with the RobotWare Add-In [StateMachine Add-In](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee), then a RAPID program is loaded by the RobotWare Add-In that provides the necessary RAPID code for this in a RAPID module called `TRobEGM` (assuming that the RobotWare option `Externally Guided Motion (689-1)` is present in your robot system).

You can then for example get/set the EGM parameters by doing something like this:
```
using namespace std;
using namespace abb;
using namespace rws;

string rapid_task = SystemConstants::RAPID::TASK_ROB_1;
RWSStateMachineInterface::EGMSettings egm_settings;

RWSStateMachineInterface rws_interface("127.0.0.1");

// Retrieve the current EGM settings
if (rws_interface.services().egm().getSettings(rapid_task, &egm_settings))
{
  cout << "Current Max Speed Deviation value: " << egm_settings.activate.max_speed_deviation.value << "\n";

  // Update the EGM settings
  egm_settings.activate.lp_filter.value = 10;
  egm_settings.activate.max_speed_deviation.value += 10;
  egm_settings.run.pos_corr_gain.value = 0.5;
  if (rws_interface.services().egm().setSettings(rapid_task, egm_settings))
  {
    // Retrieve the updated EGM settings
    if (rws_interface.services().egm().getSettings(rapid_task, &egm_settings))
    {
      cout << "Updated Max Speed Deviation value: " << egm_settings.activate.max_speed_deviation.value << "\n";
      cout << "Verify in the RAPID program\n";
    }
  }
}
```

Then you can use the following methods to start/stop EGM communication sessions (assuming the RAPID program is running):
```
rws_interface.services().egm().signalEGMStartJoint();
rws_interface.services().egm().signalEGMStartPose();
rws_interface.services().egm().signalEGMStop();
```

**Finally**, changes to EGM parameters are only applied for new EGM sessions. I.e., you will have to stop any ongoing session and restart it if you have made additional changes.

I hope this helps!

BidoMan, 2019-03-13:
That's great, this helps a lot actually, and yes i'm using the RWSStateMachineInterface in combination with the RobotWare Add-In StateMachine Add-In.

What i'm trying to do is to find the optimum control parameters using something like genetic algorithm (on irb120 robot)  that gives the best performance for the robot response for any given trajectory.

so now i will be focusing on Low Pass Filter, Poss Corr Gain and Max Speed Deviation.

jontje, 2019-03-14:
@BidoMan, I realized I forgot one aspect of the  "K-factor".

If we have:
* `PPG` = "Proportional Position Gain" (constant specified in the system configurations).
* `pcg` = "Position Correction Gain" (parameter specified during runtime for the RAPID instructions).
* `k` = "K-factor" (the resulting position gain used in the EGM control loop).

Then the "K-factor" will be `k = pcg*PPG` during runtime. Changes to `PPG` requires a restart of the robot controller, while `pcg` can be used during runtime to tune `k` by specifying "how much" of `PPG` will be used.

The `PPG` is specified in the "Default Proportional Position Gain" field of the "External Motion Interface Data" instances in the configuration:
[image]

The StateMachine Add-In's RAPID code is using the `ROB_1` instance by default, which includes filtering of EGM motion references.

Sorry for my mistake!

BidoMan, 2019-03-14:
Actually that makes much more sense now ,since the control gain should have a max range >1. Thanks for the update :+1:

gavanderhoorn, 2019-04-11:
@BidoMan: from your last comment I get the impression your question(s) ha(s)(ve) been answered.

I'm closing this issue, as it's not really something concretely actionable here.

Please feel free to keep commenting of course.

closed 2019-04-11 by gavanderhoorn