ros-industrial/abb_librws #50: I have a question about subscription

wdw753951, 2019-04-11:
I can't subscription to the "ValueChanged event" of the "RapidData" instance,when i begun to subscribed;
In the virtual controller  of 6.06,this is perfect;
But in the actual matching of version "6.01",it failed, i bought "pc interface" module from abb,but it does not work too;
when i debug the info,i find  that the web socket that used to listen to the  subscriptions is not 
set up.

jontje, 2019-04-11:
@wdw753951: If I understood you correctly, then you have RobotWare 6.01 on the real robot controller.

If this is the case, then I recommend you update the system to RobotWare 6.08 since I know that there are some issues with RWS subscriptions on earlier RobotWare versions.

Also, the RobotWare option _616-1 PC Interface_ is not needed for RWS subscriptions.

gavanderhoorn, 2019-04-11:
@jontje: even though we discussed this before, I can't remember exactly: is there a way we could implement some sort of version checking for `abb_librws`? That could avoid issues like @wdw753951 is having.

jontje, 2019-04-11:
@gavanderhoorn: Yes, that is possible, I will open an issue for it.

wdw753951, 2019-04-12:
@jontje @gavanderhoorn  Thank you both,i will update my actual robot to 6.08 to have a try ;
In fact, the rws is not stable to use in Industry,It is sometimes been disconnected , i have to restart my FlexPaint to let it can be login again,i dont know why;So i will learn the whole ros  and buy the “multiing tasks” to have a try;
Thank you very much again;

gavanderhoorn, 2019-08-26:
Closing as it seems the initial issue was resolved or at least addressed and the version checking is tracked in #51.

closed 2019-08-26 by gavanderhoorn