ros-industrial/abb_librws #64: How to handle subscription event asynchronously?

ScottYang1982, 2019-10-27:
hello,
How to handle subscription event asynchronously?
As a library could you give some application examples to show how to use the library to handle the subscription event? 
I think it should be handled when subscripted robot resources change? But how ?
Thank you very much!

jontje, 2019-10-28:
Hello @ScottYang1982, here's a rough example.

Note that you can have several RWS connections to the robot controller at the same time.

This example uses one RWS connection in the main thread, and one RWS connection in the subscription thread. The implementation of the `RWSInterface` class is currently not thread-safe, which is why I used two instances. Also, there is no "nice" way at the moment for aborting the `waitForSubscriptionEvent()` function.

```
const int NUMBER_OF_WORK_EVENTS = 5;
std::atomic_bool do_work = false;
std::atomic_int work_done = 0;

void subscriptionThread()
{
  abb::rws::RWSInterface rws_interface("127.0.0.1");
  abb::rws::RWSClient::SubscriptionResources subscription_resources;

  // Add a subscription on the controller state resource (i.e. motors on/off)
  // It is possible to add additional resources to subscribe to.
  // Note that only IO-signals and persistant RAPID variables can have the "HIGH" priority. 
  subscription_resources.add(abb::rws::SystemConstants::RWS::Resources::RW_PANEL_CTRLSTATE,
                             abb::rws::RWSClient::SubscriptionResources::MEDIUM);

  rws_interface.startSubscription(subscription_resources);

  while (work_done < NUMBER_OF_WORK_EVENTS && rws_interface.waitForSubscriptionEvent())
  {
    std::cout << "[subscription_thread] Got subscription event!\n";
    do_work = true;
  }

  rws_interface.endSubscription();
}

int main()
{
  abb::rws::RWSInterface rws_interface("127.0.0.1");
  std::thread subscription_thread(subscriptionThread);

  std::cout << "[main] Wait for subscription events...\n";

  while (work_done < NUMBER_OF_WORK_EVENTS)
  {
    if (do_work)
    {
      std::cout << "[main] Do work " << ++work_done << "/" << NUMBER_OF_WORK_EVENTS
                << " --> motors are on: " << rws_interface.isMotorOn().toString() << "\n";
      do_work = false;
    }
    std::this_thread::sleep_for(std::chrono::seconds(1));
  }

  std::cout << "[main] Finished working!\n";

  subscription_thread.join();
}
```

I hope this helps!

ScottYang1982, 2019-10-30:
Hi @jontje,
 Sure it helps a lot and I realized that for RAPID data access it seems no sign of how to access array data, am I missing anything or it's just not available to access RAPID array data? Thank u!

jontje, 2019-10-30:
Great that it helped! I have opened the new issue https://github.com/ros-industrial/abb_librws/issues/65, to reply to your follow-up question. Just so that others looking for the same thing can find it easier.

gavanderhoorn, 2020-05-20:
Assuming the initial issue / question the OP posted has been resolved/answered, I'm going to close this.

If it hasn't been resolved, feel free to comment here and we can re-open.

closed 2020-05-20 by gavanderhoorn