ros-industrial/abb_librws #71: Using RWS capabilities not currently in the interface

marip8, 2019-11-21:
I'm looking to use this library to issue commands to a controller remotely in ways that are not explicitly supported by the `RWSClient` or `RWSInterface` but which are specifiable by HTTP request (such as dynamically loading/unloading modules from the file system, setting the program pointer to certain procedures, etc.). What is the recommended way of extending the capability of this library to do actions supported by RWS that are not implemented in either abstraction class? 

I could use the `POCOClient` directly, but I would rather get something like an `RWSResult` back from these actions. My current approach makes the `parseMessage`, `checkAcceptedOutcomes`, and `evaluatePOCOResult` public members of the `RWSClient` class, posts "custom" HTTP requests using `httpPost`, and checks the result with those functions. I wonder if those functions should be static methods, separate function utilities, or if `RWSClient` should wrap HTTP posting/putting/getting and return an `RWSResult`. Or should I be going about this some other way? Any advice you could give would be helpful

jontje, 2019-11-25:
> What is the recommended way of extending the capability of this library to do actions supported by RWS that are not implemented in either abstraction class?

I have been thinking a bit about this before, but not gotten around to implement it. It would be quite useful to be able to call the RWS services directly.

> I wonder if those functions should be static methods, separate function utilities, or if `RWSClient` should wrap HTTP posting/putting/getting and return an `RWSResult`.

I would probably choose the approach to wrap the `POCOClient`'s HTTP methods in `RWSClient`. As well as making the `EvaluationConditions` struct public.

Then implement something like this in `RWSClient`:
```
RWSResult httpGet(const std::string& uri, const EvaluationConditions& evaluation_conditions)
{
  return evaluatePOCOResult(POCOClient::httpGet(uri), evaluation_conditions);
}

RWSResult httpPost(const std::string& uri, 
                   const std::string& content,
                   const EvaluationConditions& evaluation_conditions)
{
  return evaluatePOCOResult(POCOClient::httpPost(uri, content), evaluation_conditions);
}

RWSResult httpPut(const std::string& uri,
                  const std::string& content,
                  const EvaluationConditions& evaluation_conditions)
{
  return evaluatePOCOResult(POCOClient::httpPut(uri, content), evaluation_conditions);
}

RWSResult httpDelete(const std::string& uri, const EvaluationConditions& evaluation_conditions)
{
  return evaluatePOCOResult(POCOClient::httpDelete(uri), evaluation_conditions);
}
```

gavanderhoorn, 2020-05-21:
@marip8: have you given any further thought to the question you raised, or have you made any progress with what you were trying to do?

marip8, 2020-05-21:
We ended up implementing the post command in a custom class we created with this library for our project. I've been meaning to circle back around to this, but I can put together a PR for the above change in the next couple days.

gavanderhoorn, 2020-05-22:
Yes, please do.

That would be great to have here.

gavanderhoorn, 2020-08-26:
@marip8: I know how "the next few days" works ;)

Would you happen to have had a chance to look at this in the meantime?

gavanderhoorn, 2020-11-20:
@marip8: I'd still be interested in seeing the class you created.

Going to close this in the meantime, as I believe the initial question was addressed.

closed 2020-11-20 by gavanderhoorn