ros-industrial/abb_librws #72: Possibility to adjust the "level" system parameter used by Externally Guided Motion.

Mariunil, 2019-12-06:
Reading in the _Applicaiton manual -  Controller software IRC5_ I find the table below presenting an EGM system parameter called "level". As we are working towards real-time control of an ABB IRB14000 we are looking for ways to reduce latency. In this context, it would be interesting to experiment with level 0 control. 

Is it possible to configure for EGM level 0 control, if so, how can it be done?

[image]

jontje, 2019-12-09:
You specify the EGM motion level in the system configurations. E.g. in RobotStudio ->  Configuration -> Motion -> External Motion Interface Data -> Update the level of the instance you are using.

I usually have two instances specified, just for convenience. I.e. one for level 0 (raw) and one for level 1 (filtering) like this:

[image]

The instance that is used is then specified in the RAPID instruction `EGMSetup`.

**Sidenote:**
If you are using the [StateMachine Add-In](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee), then these system configurations are set up by default. And the [RWSStateMachineInterface](https://github.com/ros-industrial/abb_librws/blob/master/include/abb_librws/rws_state_machine_interface.h) class can be used to select the desired instance. For example, using level 0 (raw) can be set similar to this:
```
abb::rws::RWSStateMachineInterface::EGMSettings egm_settings;
abb::rws::RWSStateMachineInterface rws_interface("127.0.0.1");
if (rws_interface.services().egm().getSettings("T_ROB1", &egm_settings))
{
  egm_settings.setup_uc.use_filtering.value = false;
  rws_interface.services().egm().setSettings("T_ROB1", egm_settings);
}
```

gavanderhoorn, 2020-05-21:
Assuming the answer by @jontje answered the question of @Mariunil, I'm going to close this.

If not, please clarify and we can re-open.

closed 2020-05-21 by gavanderhoorn