ros-industrial/abb_librws #73: Not able to set IO Signal

jbeck28, 2020-01-07:
I'm using the sample code that jontje, specifically `a2_io_signal_node.cpp`, and I'm finding that I can correctly retrieve the current IO value, however it does not change upon calling `setIOSignal`, and a 0 is returned by the function. 

Any tips on where to look to correct this?

jontje, 2020-01-17:
- Which RobotWare version are you using?
- Are you testing against a virtual controller (i.e. RobotStudio simulation) or a real robot controller?
- Which mode is the controller in? I.e. `Auto` or `Manual`?

jbeck28, 2020-01-17:
Sorry for the lack of useful information. I'm using version 6.08.00.01 of RobotWare. I was testing on a real robot controller, and it was running in Auto mode.

jontje, 2020-01-20:
No worries, and your setup should work fine.

Can you check the access level for the IO-signal(s) in question?

**For example:**

[image]

The signal `CUSTOM_A` has access level `Default` and can't be set from a remote client, while signal `CUSTOM_B` (with access level `All`) allows remote clients to set it.

Can you provide a log of the RWS communication if the above doesn't work? E.g. the RWS interface class you use has a `getLogText()` method.

gavanderhoorn, 2020-05-20:
@jbeck28: do you still have this problem?

jbeck28, 2020-05-21:
@gavanderhoorn Sorry I never updated this, I got it working. It is critically important that the IO Signal have Access Level "All".

closed 2020-05-21 by jbeck28