ros-industrial/abb_librws #78: How to use the RWS to realize automatic loading of modules

lijianhualeigua, 2020-03-01:
Hello, 
In the configuration some instance have name attributes, so i  can use the command like this ` curl –digest -u "Default User":robotics -d "name=testinstance" -X POST  "http://localhost/rw/cfg/eio/INDUSTRIAL_NETWORK/instances?action=create-default" ` to create the default  instance the Data Params  is name, name of instance.  Then use the RWS to set the other attributes value, use the command like this ` curl –digest - u "Default User" : robotics -d "Simulated=True" - X POST "http://localhost/rw/cfg/eio/INDUSTRIAL_NETWORK/instances/testinstance?action=set" .` . But about the instance of Automatic loading of Modules in the configuration, it no have the name attributes, so even i success to create the default instance, the file attributes value is blank, the result is shown below. 

[image]

So if i want use rws to realize the  automatic load the module file features and bind the specific task,   Just like the use RobotStudio to set the configuration  instance value. what i should do.

Thank you very much!

jontje, 2020-03-03:
Hello @lijianhualeigua,

There are at least two options for doing this.

#### [Option 1]: Update the instance directly
You can set the `File` attribute if you know the `instance id`, and this can be found by inspecting [all instances in the domain type](http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/cfg_domain_type_instances.html) (e.g. `/rw/cfg/sys/CAB_TASK_MODULES/instances`).

Then you can use something like this to [update the configuration instance](http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/cfg_instance_action_set_page.html):
`curl --digest -u "Default User:robotics" -d "File=<path to file>" -X POST "http://localhost/rw/cfg/sys/CAB_TASK_MODULES/instances/<instance id>?action=set"`

#### [Option 2]: Upload a configuration file and load it

A [file can be uploaded](http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/fs_file_upload_page.html) to the robot controller and then [loaded into the configuration database](http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/cfg_system_load_page.html), and this can for example be done with `abb_librws` like this:

```
abb::rws::RWSClient rws_client("127.0.0.1");

// Create and upload a mockup RAPID system module.
std::stringstream mod;
mod << "MODULE FOO(SYSMODULE)\n"
    << "\tPROC TPWriteFoo()\n"
    << "\t\tTPWrite \"FOO\";\n"
    << "\tENDPROC\n"
    << "ENDMODULE";
rws_client.uploadFile(abb::rws::RWSClient::FileResource("foo.sys"), mod.str());

// Create and upload a CFG file (specifying automatic loading of the RAPID system module).
std::stringstream cfg;
cfg << "SYS:CFG_1.0::\n"
    << "#\n"
    << "CAB_TASK_MODULES:\n\n"
    << "\t-Task \"T_ROB1\" -File \"HOME:/foo.sys\"";
rws_client.uploadFile(abb::rws::RWSClient::FileResource("bar.cfg"), cfg.str());

// Tell the robot controller to load the CFG file.
rws_client.httpPost("/rw/cfg?action=load", "filepath=$HOME:/bar.cfg&action-type=replace");
```

This code creates and uploads a RAPID module called `foo.sys`:

```
MODULE FOO(SYSMODULE)
	PROC TPWriteFoo()
		TPWrite "FOO";
	ENDPROC
ENDMODULE
```

The code also creates and uploads a CFG file called `bar.cfg`:

```
SYS:CFG_1.0::
#
CAB_TASK_MODULES:

	-Task "T_ROB1" -File "HOME:/foo.sys"
```

Finally the code also tells the robot controller to load the uploaded CFG file. After a restart, then the RAPID module should be available in task `T_ROB1`.

#### Summary

I prefer option 2, since you don't need to dig out the `instance id` and you can set several configurations at the same time.

I hope this helps!

lijianhualeigua, 2020-03-03:
It helps me a lot.  I will to test this way in the real controller. Thank you very much for you time and help.

lijianhualeigua, 2020-03-09:
Hi,  i use the way by  rws to  upload a configuration file and load it,  i rceive the bad response, `The error message: checkout AcceptedOutCome(...): RWS reponse status not accepted`,  and the code of poco_result.poco_info.http.response.status  is 500. I find the meaning in  the HTTP Status codes,  the meaning is that INTERNAL_SERVER_ERROR(500) A generic error message, given when no more specific message is suitable.
`
RWSClient::RWSResult RWSClient::LoadCFGFile( const std::string &cfg_file_path)
{
    uri_ = Resources::RW_CFG + "?" + Queries::ACTION_LOAD;
    content_ = "filepath=" + cfg_file_path + "&action-type=add-with-reset";
    evaluation_conditions_.reset();
    evaluation_conditions_.parse_message_into_xml = false;
    evaluation_conditions_.accepted_outcomes.push_back(HTTPResponse::HTTP_NO_CONTENT);
    return evaluatePOCOResult(httpPost(uri_, content_), evaluation_conditions_);
}`
But have a amazeing thing, when i restart the IRC5 controller, the configuration parameters is loaded. What happened,  this is why?

Thank you very much!

jontje, 2020-03-10:
@lijianhualeigua,  can you provide the log text generated by the `RWSClient::getLogText(const bool verbose)` method?

And which `RobotWare` version is you using?

I just tested with `RobotWare` `6.10.01` and then I get the response `200 (OK)` instead of the response `204 (NO_CONTENT)` that is specified in the [RWS documentation](http://developercenter.robotstudio.com/blobproxy/devcenter/Robot_Web_Services/html/cfg_system_load_page.html)

lijianhualeigua, 2020-03-13:
Thanks, i just tested with RobotWare 6.07.01 and then  I  get the same result like you that response 200 (OK) instead of the response 204 (NO_CONTENT) that is specified in the RWS documentation, so it meaning that the RWS documentation content exist some mistake.

gavanderhoorn, 2020-05-20:
I'm going to close this as it seems either the issue was fixed (by going to a newer RW) or it's no longer an issue for the OP.

closed 2020-05-20 by gavanderhoorn