ros-industrial/abb_librws #80: Python wrapper?

gavanderhoorn, 2020-03-10:
Being able to use RWS from a Python application would be perfect.

I've titled this issue "Python wrapper" as that could of course be one way to do this. It may however complicate things needlessly.

Searching around, I've been able to find some mentions of using `requests` to interface with RWS, but I haven't been able to find any proper Python package.

As RWS purports to be a REST based interface: @jontje would you happen to know whether RWS comes with an API spec (OpenAPI, Swagger, RADL, RAML, RSDL, something else) which would make it possible auto-generate (Python) wrappers or runtime code generate them?

I'm not entirely up-to-date on REST and web APIs, but there are definitely more advanced (and convenient) ways of consuming a REST interface than Python `requests` and composing raw URIs.

jontje, 2020-03-10:
> As RWS purports to be a REST based interface: @jontje would you happen to know whether RWS comes with an API spec (OpenAPI, Swagger, RADL, RAML, RSDL, something else) which would make it possible auto-generate (Python) wrappers or runtime code generate them?

RWS comes in [two versions](http://developercenter.robotstudio.com/webservice):

* RWS 1.0 (`RobotWare 6`) doesn't have any such API specs (as far as I know).
* RWS 2.0 (`RobotWare 7`) is compliant with OpenAPI and Swagger can be used to generate client code, but I have not worked with `RobotWare 7` systems yet.

gavanderhoorn, 2020-03-10:
Ah, great.

Is `abb_libegm` RW7 compatible?

jontje, 2020-03-10:
> Would there be a way to make sure? :)

Of course there are ;) I can check with a colleague tomorrow, but I am quite certain there aren't any.

> Is `abb_libegm` RW7 compatible?

Did you mean `abb_libegm` or `abb_librws`?

- `abb_librws`: Unfortunately not compatible, since RWS 1.0 has been deprecated in RW7. And I haven't really started thinking about a good approach to solved this yet, mainly since I haven't worked with RW7 systems.

- `abb_libegm`: Should be compatible with RW7, but I haven't had time to test this yet.

gavanderhoorn, 2020-03-10:
Hm.

At least for RW7 we should be able to use something like [OpenAPITools/openapi-generator](https://github.com/OpenAPITools/openapi-generator).

jontje, 2020-03-11:
@gavanderhoorn, FYI, I got a no regarding auto-generation of client code for RWS 1.0.

gavanderhoorn, 2020-03-26:
Too bad, but thanks for asking.

Closing this now.

closed 2020-03-26 by gavanderhoorn