ros-industrial/abb_librws #81: Document white-listing IPs when using RobotStudio

gavanderhoorn, 2020-05-20:
Using a proxy to work-around the HTTP `500` Internal Server Error problems (#47) is OK, but it might be better to use *white-listing* of IPs which are allowed to access the RWS server instead.

While both options will allow access to the machine which is running the Virtual Controller in RS, the proxy approach essentially allows *any* remote host access, while the white-list will allow only the IPs / hosts which are listed access.

This is more fine-grained and easier to setup as well.

The general procedure is described in [Using RobotWebServices to access a remote Virtual Controller](https://forums.robotstudio.com/discussion/12082/using-robotwebservices-to-access-a-remote-virtual-controller) on the RS forum.

It would be good to duplicate the steps for this procedure here locally in this repository to avoid depending on external resources (which may go off-line at any moment).

gavanderhoorn, 2020-05-20:
Copy of the post on the RS forum:

> Hello,
> 
> To access a remote VC with RWS successfully, please perform the following steps:
> 
> On the PC which is hosting the remote VC:
> 1) If it doesn't already exist, create a file called `vcconf.xml` in `C:\Users\{user}\AppData\Roaming\ABB Industrial IT\Robotics IT\RobVC\`
> 2) Put the following content in it:
> 
> ```xml
> <?xml version="1.0" encoding="UTF-8" ?>
> <hosts>
> <host ip="10.11.125.5" />
> </hosts>
> ```
> 
> 3) Change the ip address (bold above) to the ip of the local PC from which you will be accessing the remote VC resources.
> 
> That should do it.
> 
> You're basically white-listing the IP of the client. If you don't, you'll get the "RAPI Unidentified Error" when trying to access resources on the remote VC.
>

mkhansenbot, 2024-12-23:
@joedavid-tuni - did you get this to work? I can't get it to work with RobotStudio, even following the instructions above. I still get a "RAPI Unidentified Error" when connecting. Were you able to get it working on a irc5 controller?

aidiaz, 2026-04-03:
> @joedavid-tuni - did you get this to work? I can't get it to work with RobotStudio, even following the instructions above. I still get a "RAPI Unidentified Error" when connecting. Were you able to get it working on a irc5 controller?

I was having the same issues and the proposed fix still works, tested with robotware 6.16. Make sure you are changing you file extension correctly. I had problems when creating the file through notepad and not changing the extension correctly into .xml