ros-industrial/abb_librws #84: ubuntu xenial: Invalid arguments to find_dependency

gavanderhoorn, 2020-05-22:
When attempting to build an application which depends on `abb_librws` with CMake, the build is aborted with the following error:

```
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /home/user/ros_ws/devel/share/abb_librws/abb_librwsConfig.cmake:10 (find_dependency)
  CMakeLists.txt:39 (find_package)
```

This happens on an up-to-date Ubuntu Xenial, with CMake `3.5.1`.

gavanderhoorn, 2020-05-22:
It looks like the cause is this line:

https://github.com/ros-industrial/abb_librws/blob/58ad0d361c1099cef913e8f640df25ff3358c851/cmake/abb_librwsConfig.cmake.in#L9-L10

CMake `3.5.1` does not support `REQUIRED` nor `COMPONENTS` in `find_dependency(..)`.

gavanderhoorn, 2020-05-22:
Not sure we can actually do anything about this, other than not using `find_dependency(..)` here.

gavanderhoorn, 2020-05-22:
@jontje: we may want to introduce a dummy package into the CI setup which exercises the CMake infrastructure installed as part of the `install(..)` of `abb_librws`. That would've let us know about this problem before merging #69.

jontje, 2020-05-22:
> Not sure we can actually do anything about this, other than not using `find_dependency(..)` here.

For example use `find_package` instead? Are there any pros/cons?

traversaro, 2020-05-22:
> 
> 
> > Not sure we can actually do anything about this, other than not using `find_dependency(..)` here.
> 
> For example use `find_package` instead? Are there any pros/cons?

`find_dependency` just gives a fancier error message  if a dep is not found, i.e. saying something similar to "abb_librws is not found because it was unable to find Poco" instead of just "Poco not found", so using `find_package` here is perfectly ok.