ros-industrial/abb_librws #97: Can not read system configuration data?

ScottYang1982, 2020-09-02:
HI there,
 Just like title , it's not possible to read robot system configuration data? can not find interface function.

jontje, 2020-09-02:
Hi @ScottYang1982, 

The `RWSClient` class has the [getConfigurationInstances](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/include/abb_librws/rws_client.h#L362-L370) method for reading configurations, though the return value will contain the configurations in XML format.

The `RWSInterface` class doesn't currently expose any support methods to parse the returned XML format.

I have implemented parsing, for certain configuration topics and types, in my [fork](https://github.com/jontje/abb_librws). For example, see the [getCFGArms](https://github.com/jontje/abb_librws/blob/2b6a2106cf8dfffae116129985e636ac1b166b18/src/rws_interface.cpp#L85-L123) method for configuration topic `Motion` and type `Arm`.

ScottYang1982, 2020-09-04:
Hi @jontje,
  Thanks a lot , that's very helpful!

gavanderhoorn, 2020-09-04:
Closing as it seems the reported issue has been resolved.

closed 2020-09-04 by gavanderhoorn