ros-industrial/abb_librws #98: IO signal subscription Issue

ScottYang1982, 2020-09-04:
Hi,
  When subscribe to digital output signal , if only wanna to monitor up edge(from 0 to 1) but not down edge (from 1 to 0), it seems no such option? Which means every time signal changes, from 0 to 1 and from 1 to 0, the event will trigger?

Thank you!

jontje, 2020-09-04:
Hi @ScottYang1982,

As far as I remember, then there is no way to select specific types of [RWS subscription](https://developercenter.robotstudio.com/api/rwsApi/subsrv_subscribe_page.html) events to listen for. You get a notification whenever anything happens.

However, you can inspect the events on the client side to see what occurred (i.e. by looking at the event's XML data).

Rough example checking for rising edge events:

```
abb::rws::RWSStateMachineInterface rws_interface("127.0.0.1");

// Get the current IO-signal value.
auto current_value = rws_interface.getIOSignal("SIGNAL_A");
auto previous_value = current_value;

// Start a subscription on changes of the IO-signal.
abb::rws::RWSClient::SubscriptionResources sr;
sr.addIOSignal("SIGNAL_A", abb::rws::RWSClient::SubscriptionResources::HIGH);
rws_interface.startSubscription(sr);

// Start listening for subscription events.
Poco::AutoPtr<Poco::XML::Document> p_xml_document;
while(true)
{
  if(rws_interface.waitForSubscriptionEvent(&p_xml_document))
  {
    current_value = abb::rws::xmlFindTextContent(p_xml_document, abb::rws::SystemConstants::RWS::XMLAttributes::CLASS_LVALUE);
    if (previous_value == "0" && current_value == "1")
    {
      // Rising edge event occurred
    }
    previous_value == current_value;
  }
}
```

ScottYang1982, 2020-09-05:
Hi @jontje ,
   Thanks for your quick response. Yes it's a good way to filter out the falling edge but I'm concerned about performance , for instance, if subscribing to a pulse signal with width of 20ms ,then rising and falling events triggered nearly at the same time. That will be problem ?

jontje, 2020-09-07:
> Hi @jontje ,
> Thanks for your quick response. Yes it's a good way to filter out the falling edge but I'm concerned about performance , for instance, if subscribing to a pulse signal with width of 20ms ,then rising and falling events triggered nearly at the same time. That will be problem ?

Hm, yes, I think that would be a problem for the approach above.

You can instead try setting up an interrupt on you pulse signal in the RAPID code, which inverts the value of another signal that you subscribe on via RWS. Rough example (the pusle signal is called `SIGNAL_B` here):

```
VAR intnum my_interrupt_number;

TRAP myInterruptHandler
  InvertDO SIGNAL_A;
ENDTRAP

PROC main()
  CONNECT my_interrupt_number WITH myInterruptHandler;
  ISignalDI SIGNAL_B, HIGH, my_interrupt_number;

  WHILE TRUE DO
    WaitTime 1;
  ENDWHILE
ENDPROC
```

Then in the RWS client code you should only get the rising edge events when subscribing to `SIGNAL_A`.

ScottYang1982, 2020-09-08:
Yes that's a good idea to decrease the change frequency but when trap routine is involved that means the execute time is not sure ,'cause traps are pushed in interrupt queue waiting to be executed.
I'm just curious why robot web servcices  has no feature to explicitly monitor rising or falling change.

jontje, 2020-09-08:
> Yes that's a good idea to decrease the change frequency but when trap routine is involved that means the execute time is not sure ,'cause traps are pushed in interrupt queue waiting to be executed.

That's true.

> I'm just curious why robot web servcices has no feature to explicitly monitor rising or falling change.

Have you tried the official [ABB Robotics user forum](https://forums.robotstudio.com/)? It's a better place to ask this question in my opinion, since the question is related to the RWS API.

ScottYang1982, 2020-09-16:
Hi @jontje ,
   Yes I figured it out maybe it's because if one do signal  will be pulsed or not , it's not known beforehand , so can not offer rising/falling subscription.
   But I've another critical issue:
```
_abb::rws::RWSClient::SubscriptionResources subscription_resources;
subscription_resources.addIOSignal(io_signal, abb::rws::RWSClient::SubscriptionResources::HIGH);
subscription_resources.add("/rw/rapid/execution;ctrlexecstate", abb::rws::RWSClient::SubscriptionResources::MEDIUM);_
```
if I subscribe two resources, one is io singal and other is rapid state like above code , but sometimes when I exit program and then try to run it again, it will have error of "failed to subription".
 But if tried to subscribe to two io_signal when restart my program , no such issue.
Could you try it on your computer the above code and restart test programs several times to reproduce the issue? 
Thanks!

gavanderhoorn, 2020-09-16:
@ScottYang1982 wrote:
> Yes I figured it out

please describe how you solved your problem, so we can help (future) readers avoid having to figure this out again and again.

> But I've another critical issue

Seeing as this appears to be something new and only tangentially related to your previous question, I suggest you post it as a separate issue. We increase visibility and discoverability that way.

gavanderhoorn, 2020-09-16:
As this has been resolved/worked-around, I'm going to close this.

@ScottYang1982: please do comment here with a description of how you resolved your previous issue.

closed 2020-09-16 by gavanderhoorn

ScottYang1982, 2020-09-16:
@gavanderhoorn ,
  I'm sorry that about that rising/falling change subscription , I thought I figured it out but after thinking again , my understanding is wrong, and about this question I will resort to ABB RWS team and if I got any answer will share it here.