ros-industrial/abb_librws #99: Subscribe to more than one resources with ctrlexecstate

ScottYang1982, 2020-09-16:
```
_abb::rws::RWSClient::SubscriptionResources subscription_resources;
subscription_resources.addIOSignal(io_signal, abb::rws::RWSClient::SubscriptionResources::HIGH);
subscription_resources.add("/rw/rapid/execution;ctrlexecstate", abb::rws::RWSClient::SubscriptionResources::MEDIUM);
```
if I subscribe two resources, one is io singal and other is rapid state like above code , but sometimes when I exit program and then try to run it again, it will have error of "failed to subription".
But if tried to subscribe to two io_signal when restart my program , no such issue.
Could you try it on your computer the above code and restart test programs several times to reproduce the issue?
Thanks!

jontje, 2020-09-16:
@ScottYang1982, can you check what the log says? I.e. use the RWS interface's `getLogText(...)` method after you try to start the subscription.

Sometimes I get a communication timeout event, and it can be fixed by using the `setHTTPTimeout(...)` method to override the default timeout of `400000` microseconds.

ScottYang1982, 2020-09-17:
Hi @jontje ,
```
     /**
   * \brief Static constant for the default RWS subscription timeout [microseconds].
   */
  static const Poco::Int64 DEFAULT_SUBSCRIPTION_TIMEOUT = 40e6;
```
According to the above code ,I think the default timeout is 40 seconds , which is long enough.
But Yes I still got the error _General status: EXCEPTION_POCO_TIMEOUT_ , strange.

jontje, 2020-09-17:
> ```
>      /**
>    * \brief Static constant for the default RWS subscription timeout [microseconds].
>    */
>   static const Poco::Int64 DEFAULT_SUBSCRIPTION_TIMEOUT = 40e6;
> ```
> 
> 
> According to the above code ,I think the default timeout is 40 seconds , which is long enough.
> 
> But Yes I still got the error _General status: EXCEPTION_POCO_TIMEOUT_  after calling getLogText(), strange.

[DEFAULT_SUBSCRIPTION_TIMEOUT](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/include/abb_librws/rws_client.h#L790) (40 seconds) is used for active subscriptions, otherwise [DEFAULT_HTTP_TIMEOUT](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/include/abb_librws/rws_poco_client.h#L424) (0.4 seconds) is used.

The following happens when you use [startSubscription(...)](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/src/rws_client.cpp#L459-L507):

1. Firstly, a subscription registration is requested via HTTP [here](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/src/rws_client.cpp#L483). This uses the `DEFAULT_HTTP_TIMEOUT` and is what fails in your case.
 
2. Secondly, if 1. succeeded, a `WebSocket` is created for handling the active subscription [here](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/src/rws_client.cpp#L496-L497). This uses `DEFAULT_SUBSCRIPTION_TIMEOUT`.

I hope this clarifies the issue.

ScottYang1982, 2020-09-17:
@jontje ,
 Yes you are right, I believe this is the very reason of that issue. After changing DEFAULT_HTTP_TIMEOUT from 0.4s to 1s and tested for 10 times , all succeeded.

closed 2020-09-17 by ScottYang1982