ros-industrial/abb_librws #100: How to set header ?

ScottYang1982, 2020-09-23:
Hi,
  How to set the below header when run httpPost? Do not see any setting.
```
Headers 
Content-Type=application/x-www-form-urlencoded;v=1.0

"v" represents API version, which can vary. 
```

gavanderhoorn, 2020-09-23:
Are you trying to communicate with RWS v2 servers?

jontje, 2020-09-23:
The header is set [here](https://github.com/ros-industrial/abb_librws/blob/0e2dd89fecdde4a23660a87df9f103ba0ed419a5/src/rws_poco_client.cpp#L251-L254), and according to the online documentations:

- [RWS 1.0](https://developercenter.robotstudio.com/api/rwsApi/) uses `application/x-www-form-urlencoded`

- [RWS 2.0](https://developercenter.robotstudio.com/api/RWS) use `application/x-www-form-urlencoded;v=2.0`

**[Note]:** `abb_librws` only work with `RWS 1.0`.

closed 2020-09-25 by ScottYang1982