ros-industrial/abb_librws #110: Problem with the Build of the library

kappa95, 2020-10-27:
Hi community,
I'm a newbie of ROS and i'm working on a project which involves a robot ABB Yumi  (IRB 14000) and i'm using this library: [library link](https://github.com/kth-ros-pkg/yumi). In order to control the camera i have seen that is needed the RWS library and i found this repository associated to that. So i cloned the in the src folder this package and i tried `catkin build` and `catkin_make_isolated` and i had errors like the one reported. How can i solve?
I'm using ROS Kinetic Kame with Ubuntu 16.04 Xenial.

```bash
kappa95@kappa95-VivoBook-ASUS-Laptop-X505ZA-X505ZA:~/yumi_depends_ws$ catkin build
------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/kappa95/yumi_depends_ws
------------------------------------------------------------------
Build Space:        [exists] /home/kappa95/yumi_depends_ws/build
Devel Space:        [exists] /home/kappa95/yumi_depends_ws/devel
Install Space:      [unused] /home/kappa95/yumi_depends_ws/install
Log Space:          [exists] /home/kappa95/yumi_depends_ws/logs
Source Space:       [exists] /home/kappa95/yumi_depends_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------
[build] Found '27' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> abb_irb2400_moveit_plugins                                       
Starting  >>> abb_irb2400_support                                              
Starting  >>> abb_irb5400_support                                              
Starting  >>> abb_irb6600_support                                              
Starting  >>> abb_irb6640_support                                              
Starting  >>> abb_librws                                                       
Starting  >>> gazebo_mimic                                                     
Starting  >>> industrial_deprecated                                            
Finished  <<< abb_irb6600_support                          [ 0.2 seconds ]     
Starting  >>> industrial_msgs                                                  
Finished  <<< abb_irb6640_support                          [ 0.2 seconds ]     
Starting  >>> industrial_trajectory_filters                                    
Finished  <<< abb_irb2400_moveit_plugins                   [ 0.3 seconds ]     
Starting  >>> industrial_utils                                                 
Finished  <<< gazebo_mimic                                 [ 0.2 seconds ]     
Starting  >>> yumi_cameras                                                     
Finished  <<< abb_irb2400_support                          [ 0.2 seconds ]     
Starting  >>> yumi_description                                                 
Finished  <<< industrial_deprecated                        [ 0.1 seconds ]     
Starting  >>> yumi_moveit_demos                                                
Finished  <<< abb_irb5400_support                          [ 0.2 seconds ]     
Starting  >>> yumi_support                                                     
Finished  <<< industrial_trajectory_filters                [ 0.2 seconds ]     
Finished  <<< industrial_utils                             [ 0.2 seconds ]     
Finished  <<< yumi_cameras                                 [ 0.2 seconds ]     
Finished  <<< yumi_description                             [ 0.2 seconds ]     
Starting  >>> yumi_test_controllers                                            
Starting  >>> yumi_control                                                     
Starting  >>> yumi_moveit_config                                               
Finished  <<< yumi_moveit_demos                            [ 0.2 seconds ]     
Finished  <<< industrial_msgs                              [ 0.7 seconds ]     
Starting  >>> simple_message                                                   
Finished  <<< yumi_support                                 [ 0.2 seconds ]     
Finished  <<< yumi_control                                 [ 0.2 seconds ]     
Finished  <<< yumi_test_controllers                        [ 0.2 seconds ]     
Finished  <<< yumi_moveit_config                           [ 0.2 seconds ]     
_______________________________________________________________________________
Errors     << abb_librws:make /home/kappa95/yumi_depends_ws/logs/abb_librws/build.make.000.log
In file included from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:48:0,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_client.cpp:41:
/home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:42:38: fatal error: Poco/Net/HTTPCredentials.h: File o directory non esistente
In file included from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_poco_client.cpp:43:0:
/home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:42:38: fatal error: Poco/Net/HTTPCredentials.h: File o directory non esistente
In file included from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:48:0,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_interface.cpp:40:
/home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:42:38: fatal error: Poco/Net/HTTPCredentials.h: File o directory non esistente
compilation terminated.
compilation terminated.
compilation terminated.
make[2]: *** [CMakeFiles/abb_librws.dir/src/rws_interface.cpp.o] Errore 1
make[2]: *** Attesa per i processi non terminati....
make[2]: *** [CMakeFiles/abb_librws.dir/src/rws_client.cpp.o] Errore 1
make[2]: *** [CMakeFiles/abb_librws.dir/src/rws_poco_client.cpp.o] Errore 1
make[1]: *** [CMakeFiles/abb_librws.dir/all] Errore 2
make: *** [all] Errore 2
cd /home/kappa95/yumi_depends_ws/build/abb_librws; catkin build --get-env abb_librws | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << abb_librws:make                              [ Exited with code 2 ]
Failed    <<< abb_librws                                   [ 1.7 seconds ]     
Abandoned <<< abb_irb2400_moveit_config                    [ Unrelated job failed ]
Abandoned <<< abb_irb6640_moveit_config                    [ Unrelated job failed ]
Abandoned <<< industrial_robot_client                      [ Unrelated job failed ]
Abandoned <<< industrial_robot_simulator                   [ Unrelated job failed ]
Abandoned <<< yumi_hw                                      [ Unrelated job failed ]
Abandoned <<< yumi_launch                                  [ Unrelated job failed ]
Finished  <<< simple_message                               [ 0.3 seconds ]     
[build] Summary: 18 of 25 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 6 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.1 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

gavanderhoorn, 2020-10-27:
It looks like you don't have Poco installed. Did you notice that dependency mentioned under [Dependencies](https://github.com/ros-industrial/abb_librws/tree/3946fe736672b1a9213d6213318e215d0b78036b#dependencies) in the readme?

gavanderhoorn, 2020-10-27:
> i'm working on a project which involves a robot ABB Yumi (IRB 14000) and i'm using this library: [library link](https://github.com/kth-ros-pkg/yumi). In order to control the camera i have seen that is needed the RWS library and i found this repository associated to that.

I'm actually not sure whether those packages can use this version of `abb_librws`. Have you made sure of this?

gavanderhoorn, 2020-10-27:
I'm going to assume this is a problem with missing dependencies, and as such close this for now.

If it turns out this is something else, we can re-open.

closed 2020-10-27 by gavanderhoorn

kappa95, 2020-10-27:
Thank you so much for the answer, sorry if i had posted 2 issues but github gave to me 404 and in the end it posted twice. I have checked the README and i have build and installed POCO using their documentation: 

`sudo apt-get -y update && sudo apt-get -y install git g++ make cmake libssl-dev`
`git clone -b master https://github.com/pocoproject/poco.git.`
```
cd poco
mkdir cmake-build
cd cmake-build
```
`cmake ..`
`cmake --build . --config Release`
`sudo cmake --build . --target install`

Anyway now i have similar errors anytime i'm using: `catkin_make -DCMAKE_BUILD_TYPE=Release`
Now i achieve this:

kappa95, 2020-10-27:
```bash
/usr/local/include/Poco/Exception.h:228:1: error: expected ‘;’ at end of member declaration
 POCO_DECLARE_EXCEPTION(Foundation_API, PoolOverflowException, RuntimeException)
 ^
/usr/local/include/Poco/Exception.h:228:1: error: ‘noexcept’ does not name a type
 POCO_DECLARE_EXCEPTION(Foundation_API, PoolOverflowException, RuntimeException)
 ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:97:50: error: ‘PrefixSet’ has not been declared
  void getPrefixes(const XMLString& namespaceURI, PrefixSet& prefixes) const;
                                                  ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:111:8: error: ‘XMLString’ does not name a type
  const XMLString& getURI(const XMLString& prefix) const;
        ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:153:25: error: ‘XMLString’ does not name a type
  bool processName(const XMLString& qname, XMLString& namespaceURI, XMLString& localName, bool isAttribute) const;
                         ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:153:43: error: ‘XMLString’ has not been declared
  bool processName(const XMLString& qname, XMLString& namespaceURI, XMLString& localName, bool isAttribute) const;
                                           ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:153:68: error: ‘XMLString’ has not been declared
  bool processName(const XMLString& qname, XMLString& namespaceURI, XMLString& localName, bool isAttribute) const;
                                                                    ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:174:15: error: ‘XMLString’ does not name a type
  static const XMLString XML_NAMESPACE;
               ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:175:15: error: ‘XMLString’ does not name a type
  static const XMLString XML_NAMESPACE_PREFIX;
               ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:176:15: error: ‘XMLString’ does not name a type
  static const XMLString XMLNS_NAMESPACE;
               ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:177:15: error: ‘XMLString’ does not name a type
  static const XMLString XMLNS_NAMESPACE_PREFIX;
               ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:19: error: ‘XMLString’ was not declared in this scope
  typedef std::map<XMLString, XMLString> Context;
                   ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:30: error: ‘XMLString’ was not declared in this scope
  typedef std::map<XMLString, XMLString> Context;
                              ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:39: error: template argument 1 is invalid
  typedef std::map<XMLString, XMLString> Context;
                                       ^
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:39: error: template argument 2 is invalid
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:39: error: template argument 3 is invalid
/usr/local/include/Poco/SAX/NamespaceSupport.h:183:39: error: template argument 4 is invalid
/usr/local/include/Poco/SAX/NamespaceSupport.h:188:15: error: ‘XMLString’ does not name a type
  static const XMLString EMPTY_STRING;
               ^
In file included from /usr/local/include/Poco/DOM/AbstractNode.h:22:0,
                 from /usr/local/include/Poco/DOM/AbstractContainerNode.h:22,
                 from /usr/local/include/Poco/DOM/Document.h:22,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_common.h:45,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:46,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_client.cpp:41:
/usr/local/include/Poco/DOM/Node.h:73:16: error: ‘XMLString’ does not name a type
  virtual const XMLString& nodeName() const = 0;
                ^
/usr/local/include/Poco/DOM/Node.h:76:8: error: ‘XMLString’ does not name a type
  const XMLString& nodeValue() const;
        ^
/usr/local/include/Poco/DOM/Node.h:79:16: error: ‘XMLString’ does not name a type
  virtual const XMLString& getNodeValue() const = 0;
                ^
/usr/local/include/Poco/DOM/Node.h:82:34: error: ‘XMLString’ does not name a type
  virtual void setNodeValue(const XMLString& value) = 0;
                                  ^
/usr/local/include/Poco/DOM/Node.h:187:33: error: ‘XMLString’ does not name a type
  virtual bool isSupported(const XMLString& feature, const XMLString& version) const = 0;
                                 ^
[... 2502 lines cut ...]
Failed     << abb_librws:make                              [ Exited with code 2 ]
Failed    <<< abb_librws                                   [ 2.4 seconds ]     
Abandoned <<< abb_irb2400_moveit_config                    [ Unrelated job failed ]
Abandoned <<< abb_irb6640_moveit_config                    [ Unrelated job failed ]
Abandoned <<< simple_message                               [ Unrelated job failed ]
Abandoned <<< industrial_robot_client                      [ Unrelated job failed ]
Abandoned <<< industrial_robot_simulator                   [ Unrelated job failed ]
Abandoned <<< yumi_hw                                      [ Unrelated job failed ]
Abandoned <<< yumi_moveit_config                           [ Unrelated job failed ]
Abandoned <<< yumi_launch                                  [ Unrelated job failed ]
Finished  <<< industrial_utils                             [ 1.6 seconds ]     
Finished  <<< yumi_description                             [ 1.2 seconds ]     
[... cut ...]
[build] Summary: 15 of 24 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: 8 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4.1 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

jontje, 2020-10-27:
I think the main issue is that the `Poco` version you use seems to be incompatible with `abb_librws`.

> I'm using ROS Kinetic Kame with Ubuntu 16.04 Xenial.

Since you are using `Xenial`, then you could try and run the [setup_poco_ppa.sh](https://github.com/ros-industrial/abb_librws/blob/3946fe736672b1a9213d6213318e215d0b78036b/setup_poco_ppa.sh) script to install `Poco` from binaries instead of compiling from source.

kappa95, 2020-10-27:
I did it and i have errors anyway:

```bash
/usr/local/include/Poco/Timestamp.h:269:19: error: ‘TimeVal’ in ‘class Poco::Timestamp’ does not name a type
 inline Timestamp::TimeVal Timestamp::raw() const
                   ^
In file included from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:40:0,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:48,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_state_machine_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_state_machine_interface.cpp:37:
/usr/local/include/Poco/Mutex.h:64:8: error: expected nested-name-specifier before ‘ScopedLock’
  using ScopedLock = Poco::ScopedLock<Mutex>;
        ^
/usr/local/include/Poco/Mutex.h:119:8: error: expected nested-name-specifier before ‘ScopedLock’
  using ScopedLock = Poco::ScopedLock<FastMutex>;
        ^
/usr/local/include/Poco/Mutex.h:224:8: error: expected nested-name-specifier before ‘ScopedLock’
  using ScopedLock = Poco::ScopedLock<NullMutex>;
        ^
In file included from /usr/local/include/Poco/Net/IPAddressImpl.h:23:0,
                 from /usr/local/include/Poco/Net/IPAddress.h:23,
                 from /usr/local/include/Poco/Net/SocketAddressImpl.h:23,
                 from /usr/local/include/Poco/Net/SocketAddress.h:22,
                 from /usr/local/include/Poco/Net/SocketImpl.h:23,
                 from /usr/local/include/Poco/Net/Socket.h:22,
                 from /usr/local/include/Poco/Net/StreamSocket.h:22,
                 from /usr/local/include/Poco/Net/HTTPSession.h:22,
                 from /usr/local/include/Poco/Net/HTTPClientSession.h:22,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:41,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:48,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_state_machine_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_state_machine_interface.cpp:37:
/usr/local/include/Poco/RefCountedObject.h:43:17: error: expected ‘;’ at end of member declaration
  void release() const noexcept;
                 ^
/usr/local/include/Poco/RefCountedObject.h:43:23: error: ‘noexcept’ does not name a type
  void release() const noexcept;
                       ^
/usr/local/include/Poco/RefCountedObject.h:43:23: note: C++11 ‘noexcept’ only available with -std=c++11 or -std=gnu++11
/usr/local/include/Poco/RefCountedObject.h:78:47: error: expected initializer before ‘noexcept’
 inline void RefCountedObject::release() const noexcept
                                               ^
In file included from /usr/local/include/Poco/Any.h:22:0,
                 from /usr/local/include/Poco/Format.h:22,
                 from /usr/local/include/Poco/FIFOBuffer.h:26,
                 from /usr/local/include/Poco/Net/StreamSocket.h:23,
                 from /usr/local/include/Poco/Net/HTTPSession.h:22,
                 from /usr/local/include/Poco/Net/HTTPClientSession.h:22,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_poco_client.h:41,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_client.h:48,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/include/abb_librws/rws_state_machine_interface.h:40,
                 from /home/kappa95/yumi_depends_ws/src/abb_librws/src/rws_state_machine_interface.cpp:37:
/usr/include/c++/5/typeinfo:39:37: error: expected ‘}’ before end of line
/usr/include/c++/5/typeinfo:39:37: error: expected declaration before end of line
CMakeFiles/abb_librws.dir/build.make:182: set di istruzioni per l'obiettivo "CMakeFiles/abb_librws.dir/src/rws_state_machine_interface.cpp.o" non riuscito
make[2]: *** [CMakeFiles/abb_librws.dir/src/rws_state_machine_interface.cpp.o] Errore 1
CMakeFiles/Makefile2:67: set di istruzioni per l'obiettivo "CMakeFiles/abb_librws.dir/all" non riuscito
make[1]: *** [CMakeFiles/abb_librws.dir/all] Errore 2
Makefile:127: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2
<== Failed to process package 'abb_librws': 
  Command '['/home/kappa95/yumi_depends_ws/devel_isolated/abb_irb6640_support/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/kappa95/yumi_depends_ws/build_isolated/abb_librws && /home/kappa95/yumi_depends_ws/devel_isolated/abb_irb6640_support/env.sh make -j8 -l8
```

gavanderhoorn, 2020-10-27:
High-level comment: please do not just run `sudo make install` on a system managed by a package manager. Unless you know what you are doing, it's an easy way to break things, especially when you're building software from source.

But I'd also like to repeat my question: have you made sure the `abb_librws` here in this repository is a version you can actually use with the packages from KTH? If it isn't, there is no point in trying to build them.

gavanderhoorn, 2020-10-27:
> ```
> /usr/local/include/Poco/Exception.h:228:1: error: expected ‘;’ at end of member declaration
>  POCO_DECLARE_EXCEPTION(Foundation_API, PoolOverflowException, RuntimeException)
>  ^
> /usr/local/include/Poco/Exception.h:228:1: error: ‘noexcept’ does not name a type
```

this suggests that the version of Poco you installed requires C++11, but you did not enable it while building `abb_librws`.

On Xenial, the default compiler will not enable C++11 for you, as it's too old.

I don't know in what state your system is now (with having a version of Poco in `/usr/local` and having installed from the PPA), but in a sane build environment, I'd run something like this (note: `catkin build`, as `catkin_make` cannot build `abb_librws`):

```
catkin build -DCMAKE_BUILD_TYPE=Release --cmake-args -DCMAKE_CXX_STANDARD=11
```

that *should* work.

But again: please make sure you can actually use this library.

kappa95, 2020-10-27:
> High-level comment: please do not just run `sudo make install` on a system managed by a package manager. Unless you know what you are doing, it's an easy way to break things, especially when you're [...]

i have searched for this library because in one part of the package (related to the camera of the Yumi) is needed the ABB RWS library. Infact in the CMakeLists.txt of that package there is: 
`if (abb_rws_interface_FOUND)
  message("--   ABB RWS library found!")
  message("--   yumi_cameras package will be built")
...`

gavanderhoorn, 2020-10-27:
I don't believe that's the same package:

> ```if (abb_rws_interface_FOUND) [..]```

This package is called `abb_librws`, not `abb_rws_interface`.