ros-industrial/abb_librws #123: Check Safety Status of Controller

KevinWarburton, 2020-11-06:
@marip and I are trying to make an automated state machine that uses `abb_librws` to interface with an ABB robot. I would like the state machine to handle emergency stop conditions, motion errors, and other failure modes. We have found that there is an ["errorstate" HTML endpoint](https://developercenter.robotstudio.com/api/rwsApi/ms_motionsystem_error_state_page.html) that can check for different motion error states. This is useful, but is it possible to get other safety information off of the controller (such as status of the safety circuit on the controller)? We have not found this feature within the ABB RWS documentation.

jontje, 2020-11-09:
> This is useful, but is it possible to get other safety information off of the controller (such as status of the safety circuit on the controller)?

Have you looked into configuring `System Input` actions and `System Output` statuses? For that you need to define digital input and output signals, which you then connect to input actions and output statues respectively.

For example:
- Create a digital output signal called `OUTPUT_ESTOP`, and a digital input signal called `INPUT_RESET_ESTOP`.
- Connect the `OUTPUT_ESTOP` signal to the `System Output` status `Emergency Stop`.
- Connect the `INPUT_RESET_ESTOP` signal to the `System Input` action `Reset Emergency Stop`.

This can be setup via `RobotStudio --> Controller tab --> Configuration --> I/O System`:
- `Signal` section: Add digital input and output signals.
- `System Output` section: Add connections between output signals and output statues.
- `System Input` section: Add connections between input signals and input actions.

Please see `Technical reference manual - System parameters` for more information and lists of available actions and statuses. In `RobotStudio 2020.2` the manual can be found via `File tab --> Help --> IRC5 Documentation --> System Parameters`.

Finally, you can use `abb_librws` to get and set the IO-signals.

gavanderhoorn, 2020-11-20:
I'm going to close this due to inactivity.

If you have an update @KevinWarburton please post it here.

closed 2020-11-20 by gavanderhoorn