ros-industrial/abb_librws #124: What to do when restart robot controller?

ScottYang1982, 2020-11-13:
@jontje,
  If robot controller is restarted ,then the current rws interface will lost its connection with the robot. But how to restore the connection automatically on restart of robot controller?
Thank you

jontje, 2020-11-16:
Hi @ScottYang1982,

Can you provide more context about how you use `abb_librws`?

For example, if your application is polling for data, then it should automatically connect again after the robot controller has been restarted. Assuming the polling is continued during the restart.

However, if you have, for example, set up RWS subscriptions, then they won’t be resumed after the robot controller restart (at least as far as I know). In that case you need to reregister the subscriptions.

gavanderhoorn, 2021-01-19:
I'm going to assume this is no longer a problem for you @ScottYang1982 and close this.

If you still have questions, please post a comment and we can re-open.

closed 2021-01-19 by gavanderhoorn