ros-industrial/abb_librws #126: Add missing methods, lead-through, module / program loading

dorianleveque, 2020-11-22:
Hi everyone, 

I recently worked on the robot Yumi (from ABB company) for a project of my school and I noticed during the development that some method was not implemented. In particular, I needed to control the lead-through and to be able to load a module and programme remotely to give position instructions to the arm joints using RAPID code.

By consulting the documentation I found out that it was simple to add the missing methods in the client and interface files. 
So that's what I did and I was able to test my code on a simulator (RobotStudio) and on the real robot and it works as shown in the following videos:

- test on RobotStudio: https://drive.google.com/file/d/1x9XNfOXQL9rdM93CiTaC9WfDVatNzUgk/view?usp=sharing
- movement test on real robot: https://drive.google.com/file/d/1xFfBXzxQlmXpZHcME-olONb-CCrPgHtm/view?usp=sharing
- test Lead-through on real robot: https://drive.google.com/file/d/1vlIpyDD4fowyEOJoWlTPxbcO08rPxtTc/view?usp=sharing

Can I make a pull request ?

gavanderhoorn, 2020-11-23:
I'll let @jontje comment on the utility of your proposed contributions, but from a process point of view: please open PRs, preferable 1 PR per feature (or more, if it would make reviewing easier).

It's much easier to figure out whether something would make sense to merge here if the code can be inspected, and having it as a PR immediately facilitates discussion.

If in the end it's decided to not merge a PR: no problem, we just close it.

jontje, 2020-11-23:
Hi @dorianleveque,

> Can I make a pull request ?

Yes, you are more than welcome to create a PR, but please split it up into separate PRs as @gavanderhoorn indicated.

I will be happy to review the contributions, but just so you know, then I will be fairly preoccupied with year-end deadlines and demos until at least mid-December.