ros-industrial/abb_librws #139: Setting Signal (Output) fails

simonschmeisser, 2021-04-21:
Hi

I'm trying to set a signal/output via RWS but it fails with an error message that I cannot find anything about:
```
curl --digest -u "Default User":robotics -d "lvalue=1" -d “userlog=true" -X POST "http://192.168.0.20/rw/iosystem/signals/PROFIBUS_Anybus/PB_Internal_Anybus/DO60?action=set"
<?xml version="1.0" encoding="UTF-8"?>
<html xmlns="http://www.w3.org/1999/xhtml">
<head><title>rw</title><base href="http://192.168.0.20:80/"/></head>
<body><div class="status"><span class="code">-1073414145</span>
<span class="msg">rws_resource_iosystem.cpp[2387] Reject code:-1073414145 icode:-2</span>
</div></body></html>
```
I'm running this in parallel with the "old" `abb_driver`, could those interfere?

jontje, 2021-04-22:
You can find a list of RWS return codes [here](https://developercenter.robotstudio.com/api/rwsApi/common_return_codes.html).

The description for `-1073414145` is `An error message that has not been successfully been converted to a common return code.`, which isn't so helpful.

My guess is that the issue has to do with the IO-signal's access level, so you could take a look at the system configurations.

Another approach could be to create a "virtual" IO-signal that you can set via RWS, which is then connected to the real IO-signal via a cross-connection.

simonschmeisser, 2021-04-22:
Yes, thanks, that turned out to be the problem. After lot's of googling I found a message from someone with pcsdk having the same issue. It was a long evening ;)

Given that this error message is completely generic, abb_librws propably cannot give any more detailed feedback either. Could you trigger something internally to improve that? Is there a place to document this?

The virtual IO signal worked fine. For other unlucky souls (or future me) finding this thread: Just right click in the "Signals" field and create a new signal that you don't assign to any physical hardware. Then go to cross-connections and add one that goes from your new "virtual" IO to the physical IO.

jontje, 2021-04-23:
> Given that this error message is completely generic, abb_librws propably cannot give any more detailed feedback either. Could you trigger something internally to improve that?

I will see what I can do.

> Is there a place to document this?

Previously there was a web form for reporting issues with RWS, but I can't seem to find it anymore. So I would suggest creating a post at the [official Robotics user forum](https://forums.robotstudio.com/).

jontje, 2021-04-23:
Closing since the issue has been resolved.

Please feel free to keep commenting.

closed 2021-04-23 by jontje

simonschmeisser, 2021-04-23:
[image]

I have also been unable to fill out a support request at abb.com

Sorry for spaming you with this :smile:

gavanderhoorn, 2021-04-23:
They just don't like you Simon ;)

simonschmeisser, 2021-04-23:
Aha, while this might certainly be true ... removing the **properly autoescaped** xml from the post made it go through. Now it is in manual review ... let's see if they like me there ;)