ros-industrial/abb_librws #147: Add support for RWS 2.0

herrvonregen, 2021-11-18:
Hello everyone,
a support for RWS 2.0 would be very nice to communicate with the new OmniCore controllers. I would like to work with my students on an implementation. Could you give me a small overview of the places where changes would be necessary?

gavanderhoorn, 2021-11-18:
Thanks for the offer, that's always appreciated.

We have some internal work already towards RWS 2 compatibility, but cannot release that yet.

In the meantime, the [NoMagicAi/abb_librws](https://github.com/NoMagicAi/abb_librws) fork already has RWS 2 compatibility, in addition to some other nice improvements. I would suggest to collaborate with them.

We'll look into upstreaming those changes in the future (provided NoMagicAi is interested in that of course).

gavanderhoorn, 2021-11-18:
It seems NoMagicAi has disabled the *Issues* tab on their fork (or haven't enabled it).

Perhaps @mkatliar could comment here?

mkatliar, 2021-11-18:
> We'll look into upstreaming those changes in the future (provided NoMagicAi is interested in that of course).
@gavanderhoorn of course we are interested!

mkatliar, 2021-11-18:
> It seems NoMagicAi has disabled the _Issues_ tab on their fork (or haven't enabled it).
> 
> Perhaps @mkatliar could comment here?

I have enabled issues on https://github.com/NoMagicAi/monomagic

gavanderhoorn, 2021-11-19:
> I have enabled issues on https://github.com/NoMagicAi/monomagic

is this a copy-pasta? That repository is probably private, and I assume it's not the fork we're discussing here :)

mkatliar, 2021-11-20:
Sorry, wrong link. On https://github.com/NoMagicAi/abb_librws, of course.

herrvonregen, 2021-11-24:
Hey thanks for the feedback. I just pulled the version of @mkatliar but it seems that their is some problem with the tests.
I can't build the version on the master branch. Gives me the following error:

```
/home/developer/workspace/build/abb_librws/abb_librws-test: symbol lookup error: /home/developer/workspace/build/abb_librws/abb_librws-test: undefined symbol: _ZN3abb3rws26extractDelimitedSubstringsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEccc
CMake Error at /usr/share/cmake-3.16/Modules/GoogleTestAddTests.cmake:40 (message):
  Error running test executable.

    Path: '/home/developer/workspace/build/abb_librws/abb_librws-test'
    Result: 127
    Output:
```
It just build it with `catkin build abb_librws`. Do you have any advice @mkatliar?

herrvonregen, 2021-11-24:
Nevermind.. should have cleaned the build directory beforehand..

mkatliar, 2021-11-24:
@herrvonregen we encounter the same problem from time to time, so if you find why it happens and how to fix it please submit a PR.

traversaro, 2022-01-28:
It seems that also https://github.com/JOiiNT-LAB/abb_wrapper contains a fork of `abb_librws` with support for OmniCore, even if it is not easy to say as the history of the `abb_librws` repo was not preserved.

gavanderhoorn, 2022-01-28:
Hm. It would have been nice if those were just forks ..

@gialen: what was your rationale for making disconnected copies? Collaboration is the main way we make progress, and disconnecting things like this goes against the OSS spirit.

herrvonregen, 2022-01-28:
After a few quick tests, it looks to me like the version of JOiiNT is built on an outdated version of this repository. Unfortunately, this makes it almost impossible to simply replace/use it.

pucciland95, 2023-07-29:
Hi guys,

we kept developing the repo from JOiiNT lab and it is now usable. We currently support Gofa and Yumi Single Arm with its SmartGrippers.
I basically managed to create a ros_control pkg to control the robot and link it to MoveIt!. The repo has been tested with both robots. 
I leave the link hereafter: https://github.com/MerlinLaboratory/abb_wrapper

Hoping this repo can help someone,
Cheers

pucciland95, 2024-12-04:
Hi all, I would like to know if anyone is currently working on this RWS 2.0  porting.

If not, I would like to ask @gavanderhoorn:
1) How much effort is needed to do so?
2) such porting should be done here or in a different repo?

Cheers

gavanderhoorn, 2024-12-05:
The fork mentioned in https://github.com/ros-industrial/abb_librws/issues/147#issuecomment-972779173 adds RWS 2.0 support. A number of users have been able to use it with the rest of the driver stack. I believe @herrvonregen fi.